jczic/ESP32-MPY-Jama #7: Serial port no detect

wilsonfilho405, 2023-01-23:
Hi,

I installed the Jama app on windows 10, my ESP32 with micropython is working, but when I open JAMA the com port is not detected, installed all dependencies.

jczic, 2023-01-23:
Hi @wilsonfilho405!

Thank you for your feedback. It's strange but we may be find a solution.
Can you try the Python code below and return the result please:

```python
from serial.tools import list_ports
for port in list_ports.comports():
    print(port)
```

If the correct serial or usb port is not in the result, this may be due to your configuration or a driver. If this is not the case, it will help me to find the solution on my side.
Thanks a lot :)

wilsonfilho405, 2023-01-24:
> Hi jczic,

I tested it with the code that passed and it worked, it detected the port in my case it is COM2.

jczic, 2023-01-24:
Hello and thank you @wilsonfilho405 for your answer it's very nice of you.
I am glad that it works and that you like the software. If you have any questions, remarks or ideas, please do not hesitate to tell me.

I wish you a great day!

closed 2023-01-31 by jczic