jczic/ESP32-MPY-Jama #8: WPA3 network listed as "Open" network

psykora, 2023-01-26:
Reproduced with:
WPA3 Wifi network in range
Device: ESP32C3
Micropython version: 1.19.1

### Reproduce steps
1. Click *Wi-fi Connection*
2. Look for network secured with WPA2-PSK/WPA3-SAE (OpenWRT settings)

### Actual results
The wireless network is listed as `(OPEN)` network and connection to it fails.
Connecting manually in REPL (`sta_if.connect("<network>", "<pass>")` works OK

jczic, 2023-01-26:
**Good evening @psykora and thank you for your feedback! 👍🏻** 
I'm afraid that the Wi-Fi network type identifier is zero.
Can you run this code in the IDE or terminal:
``` python
import network
w = network.WLAN(network.STA_IF)
w.active(True)
print(w.scan())
```
So, can you send me what it shows as well as the name (ssid) of your network that is in WPA3 so I can just compare and find the authentication type?
Thanks a lot!

psykora, 2023-01-30:
Hi @jczic,

sorry for the delay. There is indeed zero in the 5th element of the tuple, which probably represents the network type.
So this seems to be a micropython issue and `ESP32-MPY-Jama` can't do much until there is a new distinguishable id.

```python
[(b'WPA3Network', b'^\x02\x14\xb0\x18\xfd', 11, -35, 0, False),
 (b'OMPH2', b'\xf82\xe4\xb4S\xd8', 11, -69, 3, False),
 (b'UPC61234', b'8C}c\x05\xbc', 11, -84, 4, False)]
```

I am afraid the issue is still present in master. Will check if I have time.

jczic, 2023-01-30:
Thanks for your feedback.
Indeed since MicroPython I can't detect the WPA3 in this case.
The zero authentication type is supposed to be an open network...
I think it'll have to go up this problem :o
My software doesn't ask you for a key for the connection and that's why the connection won't work.
See you later and thanks again 👍🏻

jczic, 2023-02-09:
Hi, @psykora,
Sorry, I have just a last question on your chip :
What does this simple code returns ?
```python
import network
dir(network)
```
This is just to show if you have a dedicated constant for WPA3.
Thank you very much.

psykora, 2023-02-09:
```python
# MicroPython >>> import network
# MicroPython >>> dir(network)

['__class__', '__init__', '__name__', 'AP_IF', 'AUTH_MAX', 'AUTH_OPEN', 'AUTH_WAPI_PSK', 'AUTH_WEP',
 'AUTH_WPA2_ENTERPRISE', 'AUTH_WPA2_PSK', 'AUTH_WPA2_WPA3_PSK', 'AUTH_WPA3_PSK', 'AUTH_WPA_PSK', 'AUTH_WPA_WPA2_PSK',
 'MODE_11B', 'MODE_11G', 'MODE_11N', 'PPP', 'STAT_ASSOC_FAIL', 'STAT_BEACON_TIMEOUT', 'STAT_CONNECTING', 'STAT_GOT_IP',
 'STAT_HANDSHAKE_TIMEOUT', 'STAT_IDLE', 'STAT_NO_AP_FOUND', 'STAT_WRONG_PASSWORD', 'STA_IF', 'WLAN', 'phy_mode']

# MicroPython >>> network.AUTH_WPA3_PSK, network.AUTH_WPA2_WPA3_PSK

(6, 7)
```

Best,
Peter.

jczic, 2023-02-09:
Ok, too bad... :-/
So, I don't know why the auth type is zero when the network is in WPA3...
Strange, maybe I'll put it back together.