jczic/ESP32-MPY-Jama #9: Can connect if device running task

newmizanur, 2023-01-27:
Scenarios:
- Device is running web server using MicroWebSrv2
- If I try, Connect device then its can not connect.
- If I open using Thonny then can connect using Jama.

Solution:
- Send  KeyboardInterrupt signal like Thonny on connect then device will free to connect.

Device:
- Wemos S2 mini (I believe same issue will be for other device.)

jczic, 2023-01-27:
Hi @newmizanur and thank you for using MicroWebServ2 :)

Indeed, my software does not send a "break" when it connects.
I can add this for the next version, it's a good idea! 👍🏻 
However, do you get the error message right away or after 2 seconds?
Because if another program is already connected in "exclusive mode", your new connection will be refused immediately.

jczic, 2023-01-28:
I've fixed it : https://github.com/jczic/ESP32-MPY-Jama/commit/630b55c307b62be7a52f0b62a4de715dc01e496c
That's works (next version for binaries).
Thank you :)

closed 2023-01-31 by jczic