jczic/ESP32-MPY-Jama #10: [Feature] Add InteliSense & AutoComplete support

yarons, 2023-02-02:
A very important addition to this already wonderful tool.

yarons, 2023-02-03:
Not yet, I want to try it out with the students, we've done all the manual heavy lifting of esptool and the Arduino Labs IDE for micrpython, they are all Windows users, I'm a Linux user myself, we have some ESP32 boards laying around that we got as a donation, I'm trying to spark their imagination and see if we can do all sorts of projects together, it's a tough mission because they are not accepting my offers so far and I'm suffering from shortage of time that I can spend on what I truly love and believe in.

There are 3 points I'd like to talk to you further about (And I'm sorry for kidnapping this issue):
1. Ubuntu Package, I need to dig further into this but I'll try to see if I can help you with creating a deb/rpm files using GitHub Actions and GitHub repo so you won't need to host this, my personal need is an AUR installation script but I think it's less popular among this program's audience (and in general - I can probably help with GitHub Actions for this kind of activity, I'm also considering `docker`).
2. Localization, I think that having the ability to flash the image in my native language is very important (I can help with trying to find the right framework and setting up a localization platform like Weblate so you'll get localization volunteers from all over the world).
3. Auto detecting the board and trying to suggest an appropriate binary for it, I'm not sure it can be fully automated but semi-automated is amazing as well (It's pretty self explanatory once you look at the micropython website but it's another step that can be helpful for young students, some of my students are python developers, I'll try to see if they are willing to help with this implementation).

If it's inappropriate feel free to slap me in which way that you like and I'll try to bother you elsewhere if possible.
Thank you so much!

jczic, 2023-02-03:
Hello @yarons,
This is great! What school do you teach at? Is it in Israel? (I saw your site **hamakor**!).

Indeed, I don't know the AUR scripts at all. For the Linux version I was helped by someone about installation because personally, I have only tested the software on MacOS and Windows.
I've also never used Actions on GitHub so I don't know how it actually looks.
If you want to help, it would be with pleasure for sure :)

What do you mean by flashing the image in your language? Do you mean the default English language of the software or of the firmware image?
Yes it could be interesting to improve it in multi-languages version but for the moment it's not planned in the code (it's a bit done quickly at this level, and in raw) but everything is possible.
The thing is that when you package it in binary for Mac and Win, it becomes more complicated (for the package) but the software could propose a menu with several languages.
Yours is more complex to integrate since it is written from right to left :)
For information, the interface of the software is a Web page, so the positionings are not managed by the OS.

Once again, everything is possible 👍🏻 
Thank you very much and have a great day.

yarons, 2023-02-05:
I'm a volunteer, sadly it's not my day job :(

AUR is a personal desire, I personally believe that deb and rpm are much more important.
Explanations about GitHub Actions [RPM](https://github.com/marketplace/actions/rpm-build)/[DEB](https://github.com/marketplace/actions/build-debian-package-buster)/[AppImage](https://github.com/marketplace/actions/build-appimage/)/[Flatpak](https://github.com/marketplace/actions/flatpak-builder)/[Snap](https://github.com/marketplace/actions/snapcraft-action) builder components.
For the currently maintained builds there are [Windows](https://github.com/marketplace/actions/pyinstaller-windows) and DMG/Mac builders (The DMG builder is a little more complicated).
The advantage of using this script is that when you tag the version GitHub already builds the packages for you, there are numerous examples and you can always consult ChatGPT and you'll probably get a very solid solution for this matter.

Here is an example of building Python project with GitHub Actions (this is pip/pypi):
https://pythonprogramming.org/automatically-building-python-package-using-github-actions/

Even if it won't happen right now but you have the scripts for building the packages and you can publish those we can try and work on something and minimize the building hassle on your end and possibly develop this further to other platforms.

I mean using the UI in the respective language so that young kids will be able to flash ESP32 even without knowing English.
If it's an Electron web page I can definitely help with the RTL layout :)

Sure, no worries, thank you and sorry for the hassle, I'm at your service :)

jczic, 2023-02-06:
Hello @yarons,
I looked a bit at the GitHub actions, indeed it is interesting to automate a lot of things.
However for MacOS for example, it remains complex to get a final bundle totally viable and to encapsulate it in a custom DMG image, on Windows, it's easier but you also need to get a nice installation.
Would you like to try working on Linux deployment script (rpm) for example? That must be the easiest construction I guess.
Why not, it could be cool!

Actually, I didn't use Electron at all but Python with [pyWebView](https://pywebview.flowrl.com/) and as I'm not close to the latest web frameworks, I started with simple javascript and css from scratch.
As a result, it's quite heterogeneous as far as the default English language is concerned since there are some on the JS side but also on the Python side. We could make a json gathering all the sentences and words but that will be relatively long I think.
Especially with concatenated strings.

If I have several requests on this subject, I will indeed look into it. For the moment it's a small software that I started quickly but if it can grow, it will be with pleasure.
For the moment, the best thing would be to integrate a Linter to facilitate the coding.
Also, if you have any ideas or want to add "Jama Funcs" I'm interested too :)
See you soon.

yarons, 2023-02-06:
Hi again,
I'd like to try and build some sort of package, the most important thing is to make sure it's reproducible in terms of GitHub Actions so you can do whatever you want with it.

pyWebView is pretty awesome as well, we'll take it one step at a time but since it's web it'll be relatively convenient.

It's an awesome tool for beginners and that's all that matters :)

My 2 cents are my teaching gist which is pretty poor but it has 2 functions:
Connecting to Wi-Fi and showing the IP in the REPL and hosting a simple HTTP server.
It's right [here](https://gist.github.com/yarons/1927f990b6aa08b87c877b55cb31aff5).
I'm not sure about your standards but I think it could be a pretty cool addition if I understood correctly.
I was going to implement ping (as a receiver and blink the LED for every ICMP packet) and possibly also mdns but I need to find the time for it, hoping it's not too lame.
Thank you again :)

jczic, 2023-02-06:
Hi,
I don't know about packages on Mac and Win.
Is it possible to launch PyInstaller via GitHub actions?
For Mac the difficulty is to "notarize" the application but I saw an action that seems to be able to do it.
Yes, pywebview is a very good way to make web GUI. So it's not "native" but finally, very practical and open.
You can also make a Wifi captive-portal for your light web server no ? :))
Small Ping and DNS tools in Jama-Funcs would be really nice I think! 👍🏻

yarons, 2023-02-07:
Regarding [PyInstaller for Windows](https://github.com/marketplace/actions/pyinstaller-windows).
Not sure about Mac though.
Yes, I'll sure try that but I'm not sure about the code space though, thanks for the tips!
Thank you so much, I'll keep you updated.

jczic, 2023-02-09:
👍🏻