jczic/ESP32-MPY-Jama #12: Linux instructions gir1.2-webkit2-4.0 missing

biemster, 2023-02-08:
First of all, this is great!
When installing, I initially had issues with `WebKit2` missing. The installation instructions are very clear, but (at least for me on Ubuntu 18.04) I was missing the `gir1.2-webkit2-4.0` package.
Maybe this can be added to the README in case someone hits this in the future?

jczic, 2023-02-09:
Hi @biemster and thank you for your feedback.
We can check that with @happenpappen if there is this problem of package missing and indicate it in this case.
:)

happenpappen, 2023-02-09:
Hi @biemster,

thank you for trying the Linux version. I've created a pull requests which contains some updates regarding installation and building on Linux.

regards,

Christoph