jczic/ESP32-MPY-Jama #13: BLE Scan for ESP32 without PSRAM

biemster, 2023-02-08:
The BLE Scan Jama Func now errors out on for example a ESP32-WROOM-32 (which might be one of the more widely used ones?), with
```
MemoryError: memory allocation failed, allocating 136 bytes
```
because it does not have PSRAM on board (only about 100KB RAM free).
However, simple BLE scans like in https://gist.github.com/biemster/801c25efc1d0ca9ea28ba39781aba2aa work fine.

Is it possible to reduce the memory footprint in the Func so it runs on those low RAM ESP32s?

biemster, 2023-02-08:
I'm wondering actually how a 32KB script can fill 100KB of RAM, while the included gist runs the same BLE code but fits?

biemster, 2023-02-08:
Abolishing `import hexlify` did not seem to have any effect, as did making all the constants actually `micropython.const`

biemster, 2023-02-08:
splitting `BLE Scan.py` in two; just the Jama Func and `bleAdvReader.py` which is referenced in the source, actually makes it work. Do Jama Funcs have some support for this? I really like the idea of the Funcs!

biemster, 2023-02-09:
Also, the scanner is missing many advertisements. I'm working on something that sends an adv every 500ms, and the current Func only catches one or two during it's 7 seconds scanning cycle if it's lucky. I think the duty cycle should be optimised.

jczic, 2023-02-09:
Hello @biemster,
That's right and I'm working on this problem because it doesn't specifically concern the BLE scanner but the way JAMA-Funcs run.
Indeed, during the transmission of the code and its "real-time" execution, a lot of memory is reserved and I am looking for this problem.
I agree with you about the scanner, the base is quite simple in this little example but the messages are supposed to indicate the different data types.
I'll keep you posted (you'll be automatically notified in a future release via the software/GitHub).
Thanks to you.

jczic, 2023-02-20:
Hi @biemster, the Jama Funcs memory problem is fixed in the latest repo sources but not in the releases (binaries).
It will be in the next release. :)
(commits 3e7a2b56e3a99f958a5cf9386989ce585fdb83d5 and e5a3192802c06590767f418d31ff881e1b7f7513)

closed 2023-02-20 by jczic