jczic/ESP32-MPY-Jama #19: Ctrl+/ keyboard shortcut for commenting lines?

torntrousers, 2023-02-11:
From other IDEs I'm used to being able to comment / uncomment lines with the keyboard shortcut pessing the ctrl key and the / key. So with the cursor anywhere on a line or with multiple lines highlighted the ctrl+/ will insert a comment character at the start of the line or if the lines are already comments then it will remove the comment character.

Any chance you could add this to the code editor?

jczic, 2023-02-11:
Hi @torntrousers  and thanks for your message.
It's a good idea 👍🏻  but honestly I don't know when I can work on it  :o

torntrousers, 2023-02-11:
Hunting about I found this, which does what I'd like: https://github.com/replit/codemirror-vscode-keymap
How hard would it be to put that into this?

jczic, 2023-02-11:
Yes, this is indeed the place to put the key combinations.
However, after that, it still have to modify the editor code accordingly of course :)

torntrousers, 2023-02-12:
or this makes it look easier: https://pham.codes/blog/toggling-comments-for-custom-codemirror-modes

jczic, 2023-02-13:
Thank you @torntrousers, I saw the addon in the official codemirror addons page too:
[https://codemirror.net/5/doc/manual.html#addons](https://codemirror.net/5/doc/manual.html#addons) -> search "comment.js" on this page.
Just give me some time to add it 👍🏻