jczic/ESP32-MPY-Jama #20: Create an AppImage for this app

yarons, 2023-02-11:
I started looking at this issue.

Creating an AppImage for a Python app is slightly different than creating an app for a binary, it requires some further tinkering.

In addition I thought about adding a GitHub Actions so the AppImage will be submitted to the GitHub Packages so no other hosting service is required.

I'm guessing a snap is also an option but I'll handle that in another issue.

jczic, 2023-02-13:
@yarons: Are you talking about a MacOS AppImage then?
On MacOS it's all about packaging everything in a "bundle" according to the architecture of the platform and the AppImage is just a way to compress and present the binary.
However, with some software it is possible to set a lot of options to make it more pleasant.

yarons, 2023-02-15:
Well, there is a way to use AppImage on macOS but it's intended for Linux.
It's an app that simply launches, without installation etc., very convenient and intuitive (as a replacement for the current lengthy installation procedure).
It's an equivalent of having a single portable `.exe` file in Windows.

jczic, 2023-02-16:
Okay, so you're talking about automatically creating a binary file of the application but intended for Linux if I understand correctly? And to do it from a Mac system?
In my opinion you have to install a number of special dependencies because the binary has to be compiled by packaging a lot of things.
Do you think that Linux users prefer binaries or sources?

yarons, 2023-02-17:
@jczic AppImage is almost like a binary file but it's more convenient and cross distribution than most other implementations.
You can potentially do it on a mac but doing it on Linux (or from GitHub Actions) is easier.

AppImage handles the additional dependencies as well as it's not an actual binary but a wrapper for the application that can potentially include required extras.

Linux users can choose but since we're talking about students we should try and lower the access barrier as much as possible, at least in my opinion.
Thank you :)

jczic, 2023-02-17:
@yarons I understand now :)
Yes so, it's possible !
Ok for me.