jczic/ESP32-MPY-Jama #24: Unable to program ESP32

rdagger, 2023-02-19:
I received the following error while trying to program MicroPython firmware to an ESP32 Lolin32 Pro in Windows 10:

> Error:  This file is not a valid firmware image.

I'm using the firmware file: _esp32spiram-20220618-v1.19.1.bin_

I was able to successfully erase the chip using ESP32 MPY Jama.  I just could not program it.  I was able to program the ESP32 with the same firmware file using ESPTool.

One other small note:  The erase chip initially failed because I was connected to the board.  I had to disconnect before the erase would work.  It would be helpful to have a warning that confirms you are not connected prior to erasing or programming.

jczic, 2023-02-20:
Hi @rdagger,
So, I must to try the firmware flashing on Windows 10, it's strange but possible.
Also, you have to be sure that the chip has restarted in boot loader mode, otherwise it simply won't work. Are you sure it was in this mode? After erasing the flash, the chip must also be restarted in boot loader mode.

For your idea to prevent the user, yes it's needed! 👍🏻

rdagger, 2023-02-20:
@jczic I've never had to put any of my ESP32's into bootloader mode to erase or flash the MicroPython firmware (except for some of the newer S2 & S3 versions.)

ESP32-MPY-Jama had no trouble erasing the board.  Also, the error message seems to indicate a problem with the actual firmware file as opposed to the state of the board.

jczic, 2023-02-20:
Ok @rdagger, do you have the msg "This file is not a valid firmware image" ?
What is your version of esptool ? (It's indicated on the top of firmware's page).
It doesn't seem to work with a version a little older than 4.4. I must see that...

rdagger, 2023-02-20:
Yes, that's the message.
Version 4.2.1

I upgraded to 4.5 and it works now.  Thanks!

jczic, 2023-02-20:
So I have to display an information to the users in case of too old version.
However, I'll have to test all this! **:)**

closed 2023-02-20 by jczic