jczic/ESP32-MPY-Jama #26: macOS keyboard input goes to terminal, not App

guineau, 2023-02-27:
On macOS, when I run from the local repo clone (python src/app.py) all the input from the keyboard continues to go to the terminal where I started the app, and not to the app itself. So for example I try to type a password for the WiFi, it shows in the terminal.

This does not happen in the app when installed from the DMG. I also tried creating a binary (create_binary.sh) but the same issue occurs.

How do I get around this (to use the latest source) or how can I make the .dmg installer version?

The latest source connects to the device super fast!

thanks!

jczic, 2023-02-27:
Hello @guineau and thanks for using ESP32 MPY-Jama :)
On MacOS, indeed, the launch from a terminal makes that the keyboard events are received in this same terminal.
I advise you to go in the sources directory with the **Finder**, to right-click (two fingers on the Trackpad) on the **app.py** application and then to open it with **Python Launcher** (which is installed with the MacOS version of the official Python).
Thus, the python that is executed is in GUI mode and there is no terminal so the software works correctly.
It's ok for you?

guineau, 2023-02-27:
I can't find any Python Launcher (not in open-with etc.)

I'm running macOS Ventura.

john

jczic, 2023-02-28:
@guineau, the last Python pkg for MacOS :
https://www.python.org/ftp/python/3.11.2/python-3.11.2-macos11.pkg

Available on https://www.python.org/downloads/release/python-3112

guineau, 2023-03-01:
Interesting. I installed that Python you linked to. Then used that python to create a venv in a terminal. And I can now run app.py from the terminal and input goes tot he app and not the term.

Must be something with the python install? I 

￼

jczic, 2023-03-03:
Hi @guineau.
It must be configured with **Cocoa** or **QT** to intercept the graphical elements correctly.
It is clear that Python must receive the events of the MacOS UI on its own application window.
In any case it works for you and that's the main thing.
Don't hesitate to come back to me if you have other problems :)

guineau, 2023-03-03:
Thank you Jean-Christophe!JohnSent from my iPhone

closed 2023-03-03 by jczic