jczic/ESP32-MPY-Jama #29: Great tool

ghvau, 2023-03-24:
Why can i see UART Test twice times inside Jama Funcs ?

jczic, 2023-03-25:
Hi @ghvau, what OS are you on and is it the packaged binary version of the software or the sources directly from the GitHub repository?
Also, could you have exported it and then accidentally imported it (try to remove them)?
Otherwise, I have no idea, it remains after the software is relaunched?

ghvau, 2023-03-25:
Hi Jean,

OS: Ubuntu 22.04
Installation: https://github.com/jczic/ESP32-MPY-Jama
Running: Source code

Testing:

Jama Funcs menu: _unsorted_ functions (with 2 UART Test), _incl._ the 
function template!

File dir: functions w/o function template, the template was in the upper 
dir!

Empty the complete function file dir.

Restart jama / Jama Funcs shows 1 UART Test and the function template !?

Remove both functions inside Jama Funcs.

After restore the complete file dir and restart jama / Jama Func shows 
the _sorted_ functions and in right quantity !

Problem solved ??!!!

jczic, 2023-03-26:
Hello @ghvau, let's just say that I have no idea how this problem could have arisen.
The system will search in the folder linked to its resources (in **".../content/Jama Funcs/*"**) but also in the functions that you import and that you can therefore delete, under Linux in **"~/.config/ESP32 MPY-Jama/Jama Funcs/*"**.
Existing functions in the software cannot be deleted.

Your concern about the unsorted list with 2x UART is quite complicated to understand. Maybe a crash in the middle of the system reading...?
The appearance of the template in the middle seems impossible...

I'm glad you were able to solve this problem 😊, I haven't had any other feedback about this on Linux yet.

jczic, 2023-03-26:
@ghvau, do you think you can close this issue?

closed 2023-03-27 by ghvau