jczic/ESP32-MPY-Jama #30: Getting script by download or export failed

ghvau, 2023-03-25:
I play a little bit with the jama-gui and get the following situations:

1. Creation mpy script, store it on device, can run it in the expected way, but the DOWNLOAD function general get message: "An error has occurred" 
2. Use different Jama Funcs as expected, but the EXPORT function general get message: "An error occurred while exporting this function"

Ubuntu 22.04, source code running (github), ESP32+CP2102

jczic, 2023-03-26:
Hello @ghvau and thank you for using the soft and for your comment!
I understand that the software opens a dialog box to "load" or "save" a file is a problem for you.
In my opinion it is a problem of access rights to the default folder on which the dialog box opens.
Try to launch the software with more rights? It is probably too restricted.

ghvau, 2023-03-26:
Hi Jean,

after some "chmod's" w/o succsess i do the extreme test and run the 
program as ROOT: -> same behavior.

It's not a problem, it is unlovely. But i have a workaround: copy and 
paste :-)

jczic, 2023-03-26:
@ghvau, maybe I found the cause of the rights problem for the default folder on your Linux.
Just telle me, do you see the **save dialog box** correctly if you click on the **"this template"** button in the **"Jama Funcs"** page?

ghvau, 2023-03-26:
@ghvau <https://github.com/ghvau>,

see private mail.

The "copyfile" function crashed because of different parameter type.

Different "string" handling in Window/Linux ???

jczic, 2023-03-26:
I did not receive any particular private email?
Strings are used in **copyfile** function.

To tell the truth, I never tested it directly under Linux but maybe @happenpappen can help us?

It is strange that it plants in this place.
Does the modification I made allow you to see all the dialogs at least?

ghvau, 2023-03-26:
... send private mail again !

ghvau, 2023-03-26:
... "Does the modification I made allow you to see all the dialogs at least? .."
For me, there is no change inside the EXPORT dialog and function.

Maybe there is a different dialog depending used os ?

My sended modification (string modification inside Export Jama Funcs) works for me.
Maybe there is an equal situation inside DOWNLOAD Code Editor.

jczic, 2023-03-26:
Ok, I seen your mail and I fixed that, thank you!
I've just used "str()" to transtype the objc -> https://github.com/jczic/ESP32-MPY-Jama/commit/340b421112e1d2755fd0be72e67a37955517281a
Can you retry with latest sources please?
Thank you again 👍🏻

ghvau, 2023-03-27:
I am sorry but this "easy" type conversion don't work, i get the same error/situation. I try it also before i found the other solution, maybe not the best way, but working in Ubuntu environment.

a =  ('/', 'h', 'o', 'm', 'e', '/', 'n', 'o', 'd', 'e', 'm', 'c', 'u', '/')
print(a)
('/', 'h', 'o', 'm', 'e', '/', 'n', 'o', 'd', 'e', 'm', 'c', 'u', '/')
print(str(a))
('/', 'h', 'o', 'm', 'e', '/', 'n', 'o', 'd', 'e', 'm', 'c', 'u', '/')
an=''.join(str(x) for x in a)
print(an)
/home/nodemcu/

ghvau, 2023-03-27:
This will be the smallest modification, which run in my env:
copyfile(filename, ''.join(r))

ghvau, 2023-03-27:
A other approach maybe to find out, why the **target file** name for disk writing was a type "tuple" and not "string" ?!

jczic, 2023-03-27:
Ok @ghvau  it's done with **join()**.
Indeed, it's very strange that the return of the file dialogs is a Tuple on your Linux distribution. I have to report this to the pywebview developer.
So the source update should work on you now.
Thanks to you. 👍🏻

jczic, 2023-03-27:
Thanks for your interest in this software, I hope you like it and that it displays well on your OS (animations etc) 😊

happenpappen, 2023-03-28:
Is this resolved or does it need further testing? I could look after this this evening, yesterday has been quite busy 😁 .

jczic, 2023-03-28:
Hello @happenpappen, yes it's ok thanks.
Have you been able to test the latest features?

happenpappen, 2023-03-29:
> Hello @happenpappen, yes it's ok thanks. Have you been able to test the latest features?

Yesterday evening i fetched the current version and tried it using a Seeed Studio XIAO ESP32C3 which worked very good. I tried to step through all major functions and did not encounter any errors. Do you have a specific feature in mind which needs testing?

jczic, 2023-03-30:
Hi @happenpappen, that's great, thanks a lot 👍🏻 
There are mostly new features like last Jama funcs, SD card, Ethernet connection and now I just updated a new feature to allow to retrieve the state of some configurations automatically at boot (autonomously in the chip)
:)