jczic/ESP32-MPY-Jama #33: Open ttyUSB device in Ubuntu (list pos empty)

ghvau, 2023-03-31:
[image]

Inside device selection there is an EMPTY position which represent the connected device.
dmesg: usb 2-1.2: cp210x converter now attached to ttyUSB0
If selected the device will connected and is usable.

Ubuntu 22.04
Jama just downloaded

jczic, 2023-03-31:
Hi @ghvau, do you have download from binaries or from repository sources?
I'll check that.

jczic, 2023-03-31:
Ok that's works thanks

ghvau, 2023-03-31:
Problem was size or character inside port description.

See private mail

jczic, 2023-03-31:
Well @ghvau, it's fixed on commit : https://github.com/jczic/ESP32-MPY-Jama/commit/9603f12a1870eee4a723d3d990ca495f267d57e4
:)

ghvau, 2023-04-01:
> Well @ghvau <https://github.com/ghvau>, it's fixed on commit : 9603f12

Thanks, solved !

closed 2023-04-01 by jczic