jczic/ESP32-MPY-Jama #34: Boot configuration (some wifi connection) get problems (Ubuntu)

ghvau, 2023-03-31:
[image]

Router1: b'NG-AP-ST', b'P\xe66\xc1\x0e\x10', 1, -71, 7, False 
Router2: b'Freifunk', b'\xd2\x95\x05I\xd48', 9, -86, 0, False

Testsequence:

I start with empty Boot config:
[image]

Set up connection Router2 via Network or Connect Wi-Fi:
Store icon appears (rotate clockwise) but nothing happens/stored !
The store icon stays permanent with clockwise move.
But connection established (can get NTP)
[image]

I close the connection of Router2 and setup Router1
The Store icon appears and data can be stored !!!
[image]

After successful storing I can also (after close R1 and new setup Router2) store data ?!
But it seems be not the data of Router2.

When I start jamaBootCfg manually I can see always “Connecting Wi-Fi to "NG-AP-ST"... OK”,

ghvau, 2023-03-31:
Ubuntu 22.04, github source code

jczic, 2023-03-31:
Hi @ghvau and thank you for your feedback!
It was a bug and I fixed it right then and there.
It works now, thanks again. 👍🏻

ghvau, 2023-04-01:
Thanks, solved!

closed 2023-04-01 by jczic