jczic/ESP32-MPY-Jama #35: Neon (Ubuntu) blanks screens with pywebviewqt (but works with gtk)

wd5gnr, 2023-03-31:
I consistently get two blank screens and a hang when trying to run under KDE Neon (basically Ubuntu). I tried a few things. To isolate the problem I did this:

```
import webview

def main():
    w=webview.create_window("Wow",html="<B>OK!</B><HR><I>Yay!</I>")
    webview.start(gui='qt')

if __name__ == "__main__":
    main()
```

If I am not in the venv and I have python3-webview installed it works. Without that package, of course, it doesn't.

If I go into the Jama venv, then it does not work (same as the app) with or without python3-webview (but with the pip-managed webview[qt].

However, if I do: pip3 install pywebview[gtk] and then modify the code to include gui='gtk' on the start  line it works. Making the same change in app.py allows MPY-Jama to run also.

jczic, 2023-03-31:
Hi @wd5gnr, I really appreciate your feedback 👍🏻  and I will try to apply this solution for Linux. However, I would like to point out that other users have been able to run the software successfully under Linux and QT, so perhaps this solution only applies in certain particular cases such as KDE Neon. I will definitely take a closer look at this.
Thank you again for sharing your experience with me!

@happenpappen, if you have time, as someone who also uses the software on Linux, do you have any insights on this matter? Thanks a lot :)

wd5gnr, 2023-03-31:
Yes both of those show satisfied. There is evidence that it isn't just me and the test.py shows it isn't you, it is some setup of pywebview[qt]:

https://github.com/r0x0r/pywebview/issues/929

I had not noticed the 890 one you mentioned ... will look at that.

jczic, 2023-04-01:
@wd5gnr and @happenpappen, what do you recommend to solve this issue for sure, given that I don't have a Linux to test it myself?
Is it necessary to always force the use of **GTK** under Linux?
Is there anything that needs to be added or modified in the Linux section of the **repository readme**?
Thank you very much for your help!

wd5gnr, 2023-04-01:
Without knowing the cause, I would hesitate to make too many big changes. Here's what I would propose:

In conf.py add a variable for webviewGUI:

'''
WEBVIEW_GUI = ''
'''

then in Start (app.py) I would say

...
if conf.WEBVIEW_GUI = '':
    webview.start([stuff that is there now])
else:
    webview.start([stuff that is there now],gui=config.WEBVIEW_GUI)
'''

That's my suggestion

wd5gnr, 2023-04-01:
Oh and then, of course, in the Linux readme, you might point out that change conf.py to set conf.WEBVIEW_GUI to 'gtk' is known to help instances where qt is broken and that other values are qt, and the others that can go there (I don't remember the one for Windows).

happenpappen, 2023-04-01:
@jczic : I tried reproducing this on my laptop (which runs on Ubuntu 22.04 as well) by installing the Neon desktop, but ended in a mess regarding the installed packages. If i had sufficent ressources available on my laptop, i could try this in a virtual machine, but that's currently not possible. 

I like @wd5gnr idea of making this a configuration option which has a safe default setting. Maybe it is possible to check the available engines on the first startup of the program and set the config option automatically?

wd5gnr, 2023-04-01:
Actually, if you want to have it where the config is None to take the default, here's how to do it (had to do some reading):

```
from functools import wraps

def remove_none_from_kwargs(func):
    @wraps(func)
    def wrapper(*args, **kwargs):
        func(*args, **{k: v for k, v in kwargs.items() if v is not None})
    return wrapper

def test(arg=100):
    print(arg)
    
test=remove_none_from_kwargs(test)
test(arg=None)
test(arg=33)
```
 
So using a scheme like that you could pass None to start (probably need to add a self to wrapper) and get the default, but if you pass, say, 'gtk', you'd get that. I'd make the change and send a PR but it would be next week before I could do that so if you want to just grab it from here that's perfectly fine. You should probably make the call of exactly how far down this rabbit hole you want to go anyway ;)

jczic, 2023-04-01:
:+1: Thank you both so much, I will see what I can do quickly, but tonight, for this option!
I'll add the doc in the readme as well and tell you here when done :)
Thanks again, it's good that it's moving in the right direction!
See you soon.

jczic, 2023-04-01:
Just @wd5gnr, if it does not work, you see 2 white screens but what exactly is going on?
Does it crash the software and it close or do the 2 screens stay open indefinitely?
Maybe a **try...except** on **start(...)** is raised?

( I would like to be able to detect the problem automatically because changing a configuration in the sources implies a desynchronization with the repository. Or maybe run app.py with an argument... )

wd5gnr, 2023-04-01:
You get two blank white screens and then they're hung up so they're not
processing events. The only way I can stop it is to externally, kill it or
press control z to put it to sleep and then I can kill it while it's asleep
but clicking on the x or even pressing control. C does not cause it to exit
when it gets into that mode.

I ran it through the debugger and essentially webview start. Just never
enters the event loop. It's getting hung somewhere in there.

jczic, 2023-04-01:
Ok I see, it does not receive the WebSocket (there is my little HTTP/WebSockets server inside).
Normally, the second window is hidden at startup and the first one is the splash-screen but @happenpappen modified line 67 to not hide the second window anymore under Linux (maybe it didn't work) :
```python
hidden = not (conf.IS_WIN32 | conf.IS_LINUX),
```
If you look at the side of this line, you will see the creation of the 2 UI windows.

So the program does not crash and there would be a way to automatically detect it because the WebSocket does not connect after a certain time between the browser and the internal server.

Also, you can pass ```gui=None``` without problem to ```start()```, I tested and pywebview takes the first best browser in a certain order (see https://pywebview.flowrl.com/guide/renderer.html).

Everything of pywebview arguments is here:
https://pywebview.flowrl.com/guide/api.html#webview-start

wd5gnr, 2023-04-01:
I thought that as well,   but  I was able to open a browser on the server socket mentioned  in the config file. So I don't think it is that simple. The fact  that it works with gtk tells me it is something with the qt rendering. The fact that I can make it work without the venv and my test case doesn't work with the venv tells me it is some sort of setup/version mismatch....

jczic, 2023-04-02:
@wd5gnr,
I added the management of an argument allowing to force the web engine.
Try to run the program normally (when it crashes) and it should tell you how to do it.
Then test with the gtk argument to see if it works as expected :)
Thank you!

jczic, 2023-04-06:
Hi @wd5gnr, is-it fixed for you?

wd5gnr, 2023-04-06:
I'm not sure what you are expecting to happen. The result for me is the same. I launch the code and I get what appears now to be a single window in the same state as before: blank and hung up. From the command line, the only thing I can do to get control back is Control+Z. Then I can kill the process. Running it again does nothing. There are no --help options.

So... no, unless I don't understand what to do, it is not working. I did not look at your commit, but I will do so now.

wd5gnr, 2023-04-06:
Ok -g gtk does work. But the code to detect it is hung isn't working. I'm not surprised because, well, it is hung ;-) Everything is blocked.

jczic, 2023-04-06:
Hmm :(
I think it must be a blocking when launching the interface with GTK or that all the javascript on the interface crashes completely because the blocking detection code (line 1560) is not executed under Linux (no splash screen).
If you see the UI, it's because it passed the startup otherwise, if you still have a white window, it's because there is a problem when loading GTK with pywebview.
For the moment, I don't see where this problem could be located at my level and it's not easy to look for it (I don't have any Linux for test...).

ghvau, 2023-04-06:
Hi @[jczic](https://github.com/jczic), [@wd5gnr](https://github.com/wd5gnr),

some info’s from me as Ubuntu 22.04, Gnome user, it’s only info !!!

**First of all, its work for me as described in the Jama Linux installation part of the readme!**

When i install Jama the first time and i see the name of the needed software “...qt…” and I thought i’m running QT!
But during the dialog “Args force GUI” i am surprised that I really use GTK!  

I can start Jama (like readme) w/o parameter and after the fix “Args force GUI” also with “-g gtk” !

**If i try to run Jama with “-g qt” i get still the same here described situation (white hang window)!**

I can see the following GTK processes:
[image]

and QT processes
[image]

Detail off the 4 QTWebEnginePro:
1.
[image]

2.
[image]

3.
[image]

4.
[image]

It must be a problem in cooperation gtk/qt.
Maybe you can see something.

wd5gnr, 2023-04-06:
I agree it is nothing you are doing that is causing this because even my simple test program doesn't work without forcing gtk.Like I say, when you single step, the call to start on the webview is what hangs. I haven't traced further than that because it is upstream. But there are enough people using KDE that I think along with the switch just a note in the "how to run for Linux" would be sufficient and maybe one day the upstream will get it fixed.

jczic, 2023-04-07:
In reality, the pyWebView library, which uses the best web browser available, decides to use QT or GTK depending on what it finds on your system.
However, I added the possibility to force it to start with QT or GTK as an argument.
On the other hand, when you force QT or GTK in this way, it will start with the first one, but if that doesn't work, it will try the other one anyway.

ghvau, 2023-04-08:
From my point of view: nothing. It's only the try to confirm, that I can reproduce this situation!

jczic, 2023-04-08:
I added the info to force the GUI in the readme, commit: https://github.com/jczic/ESP32-MPY-Jama/commit/937230d63a0daf1b5647aeb94ff07cb687a34dbc
It's ok for you? Can I close this issue?

wd5gnr, 2023-04-08:
> I added the info to force the GUI in the readme, commit: 937230d

That's fine with me.

Thanks for working this.