jczic/ESP32-MPY-Jama #36: "Update startup details" don't work (Ubuntu)

ghvau, 2023-04-03:
The last Update [Update startup details] don't work on Ubuntu 22.04 / source code
After start a short flash up of "gears window" and nothing else.

jczic, 2023-04-03:
Hi @ghvau, it's a delay issue I think.
Could you increase the ```START_GUI_AND_WS_TIMEOUT_SEC``` in conf.py (7 seconds by default) and retry?

For nothing has changed except the ability to force the use of a GUI engine but also to detect when an engine does not start.
And apparently, in your case, this last point passes the 7 second time limit and the application decides to close itself by indicating the error message.

jczic, 2023-04-03:
@ghvau, thank you for your tests 👍🏻 
So it closes after 15 seconds?
Does it do the same thing if you completely remove the line ```gui = forceGUI``` in the function ```Start(...)``` line **1607** of the **app.py**.

I also just changed a sleep to an async sleep during  startup, maybe this was causing a problem on your system.

It's annoying because I don't have a Linux like yours to test :(

ghvau, 2023-04-03:
If you start the app with parameter -g qt you see

[image]

It looks equal, also the gtk modul was started ?

jczic, 2023-04-03:
Thanks for your feedback.
You mean that with the line gui=forceGUI and without passing any parameter it doesn't work while using GTK?
That would mean that setting ```gui=None``` would be different from not setting it at all or that gui would be different from None.

Could you possibly add the line ```print(forceGUI)``` just before the call to Start(...) to know what it is worth when passed?

Also, do you think you could look at what version of the **pywebview** lib is used in your environment?
I'm having trouble understanding as I don't have anything to reproduce this problem :(
Thank you very much!!

ghvau, 2023-04-03:
print(forceGUI)
W/o parameter -> None
With "-g qt" -> qt
---------------------------------------------------------------
pywebview[qt]
Requirement already satisfied: pywebview[qt] in ./venv/lib/python3.10/site-packages (4.0.2)
Requirement already satisfied: bottle in ./venv/lib/python3.10/site-packages (from pywebview[qt]) (0.12.25)
Requirement already satisfied: proxy-tools in ./venv/lib/python3.10/site-packages (from pywebview[qt]) (0.1.0)
Requirement already satisfied: PyQt5 in ./venv/lib/python3.10/site-packages (from pywebview[qt]) (5.15.9)
Requirement already satisfied: pyqtwebengine in ./venv/lib/python3.10/site-packages (from pywebview[qt]) (5.15.6)
Requirement already satisfied: QtPy in ./venv/lib/python3.10/site-packages (from pywebview[qt]) (2.3.0)
Requirement already satisfied: PyQt5-Qt5>=5.15.2 in ./venv/lib/python3.10/site-packages (from PyQt5->pywebview[qt]) (5.15.2)
Requirement already satisfied: PyQt5-sip<13,>=12.11 in ./venv/lib/python3.10/site-packages (from PyQt5->pywebview[qt]) (12.11.1)
Requirement already satisfied: PyQtWebEngine-Qt5>=5.15.0 in ./venv/lib/python3.10/site-packages (from pyqtwebengine->pywebview[qt]) (5.15.2)
Requirement already satisfied: packaging in ./venv/lib/python3.10/site-packages (from QtPy->pywebview[qt]) (23.0)
--------------------------------------------------------------

Set PYWEBVIEW_GUI direct before webview.start and use it with :

- Jama start w/o parameter
PYWEBVIEW_GUI:  qt
forceGUI:  None

- Jama start with "-g qt"
PYWEBVIEW_GUI:  qt
forceGUI:  qt

-----------------------------------------------------------------------------

There is no change in the error situation or error description  !!!

I have also the lastest Jama version which runs for me (version -> fix problem with size of port description). If i start this version, i get also the same traceview as above, means also a module gtk.py starts, but Jama works ?!?!

jczic, 2023-04-03:
Ok I see, GTK is still launched after QT in case QT does not work.
So if it doesn't change anything for you on the latest version then I think I know where the problem comes from, I changed something else that no longer hides the software main window at startup.
Maybe it works, **line 69** :
Instead of `hidden = True` you can try `hidden = not conf.IS_LINUX`
This bug also existed under Windows but has since been fixed. It must persist under Linux then...

ghvau, 2023-04-04:
If I start Jama with "-g qt" i get two (small/big) white empty windows who disappear after a few sec.

BUT if i start w/o parameter (like the past), **you have found the half way for the solution**, i get the dialog mask of jama and i can select the items.
BUT: jama starts always with reduced window (also in the past!). If i use this reduced size jama will be closed after a few sec.  
BUT: If i switch after start directly to a full size jama window 
[image]
, jama stays stable !!!
**During the switch** to full size window there is flash up a message/info/error(?) on the screen, but its to quick, i cant read it.

jczic, 2023-04-04:
Ahh! Thanks for the feedback @ghvau! 👍🏻 

First of all, don't use the QT parameter because you actually have to run the program in GTK mode on your Linux.
With the `g -qt` arguments: the UI looks for QT first and then GTK.
With the arguments `g -gtk` : the IU looks first for GTK and then for QT.

As for the windows, there are indeed 2 of them:
The small square window is in fact a splash screen which is supposed to appear during the loading time and then the big window is the software.
Normally, the big one is hidden at the beginning, the time that all the interface loads...
Apparently there is a problem at this level and all I have restored is to display the main window right away. The small one remains. That's why you see 2 of them.
Concerning the message and the fact that you have to click on the main window and not the small one, I think it's a problem of focus on the objects/windows that stays somewhere and makes the interface crash.

Also, there is a 15 second delay during the launch. After this delay, the software closes automatically if everything has not started correctly, by writing the message in the terminal about the arguments.

For the UI message that you don't have time to see, I have no idea what it is.

However, I wonder why it worked better before now.
Tell me, how long (seconds) does it take for the software to launch on your system?
And also, after how long does it close when you don't click on the main window?

Thank you very much for your feedback, we are making progress, it will work properly soon 😎

ghvau, 2023-04-04:
- Start up (launch of jama) was very quick ~ 2 sec.
- Close window after ~ 12 sec.
- Reduced window means small window frame size (controlled by the 3 window frame icon). In reduced size the jama window closed independent of with or w/o focus. The window was visible every time.

There is a other point which i ignored in the past (no handicap), but it is still there. If i go after start (reduced window) in Jama Funcs i can see only the header line and the 3 lines of description **but no functions**! After switching to full size window the contents changed and the functions were shown/appears.

jczic, 2023-04-04:
I just updated the sources by removing the splash screen under Linux!

ghvau, 2023-04-04:
**WOW ..., congratulations ..., that's it !!!**

Ready for the Ubuntu future!

After a short basic test all points seems to be solved, include the missing overview (in small window frame) of the Jama Funcs scripts.

Thanks!

jczic, 2023-04-04:
Excellent! 🎉  **thanks** @ghvau for all feedbacks and for helping me solve this problem :)
So I will compile the binaries for Mac&Win and update the readme to offer this new version!
Don't hesitate to let me know how you use it if you can.
Thanks again and see you soon.