jczic/ESP32-MPY-Jama #37: Getting a timeout error when uploading boot.py

nio101, 2023-04-08:
Thanks for the app, it's great, but I'm getting this error and I can't upload or save boot.py on my device :
[image]

Uploading other files works.

jczic, 2023-04-08:
Hello @nio101 and thank you for using this software :)
It's strange that only this file doesn't want to upload.
I see that you are on a **LOLIN S2 MINI** board and that it has a MicroPython firmware by default.
Can you make a screenshot of the "System" page in the tool, so I can see the versions please?
Maybe the `boot.py` file is protected...
Also, what happens if you open it with the IDE and save it with **Ctrl+S** instead of using the "Upload" button?

nio101, 2023-04-08:
Here is the system info :
[image]
When trying to save directly, I get the 'an error occured' message...
I can upload a boot.py using other tools (thonny for example).

jczic, 2023-04-08:
Thank you for feedback, indeed the versions seem good, why only `boot.py` and not the others...
Maybe a problem of opening the local file and not the one on the device.

Can you try the 2 commands below in the terminal to see what it returns on your side?
```python
import os; os.stat('/boot.py')
```
and
```python
with open('/boot.py', 'rb') as f : print(f.readlines())
```

nio101, 2023-04-08:
The result is : '(32768, 0, 0, 0, 0, 0, 3359, 734295654, 734295654, 734295654)' for the first command.
The second command print successfully each line of boot.py.
I have the same issue with the ubuntu version.

jczic, 2023-04-08:
Ok thanks, I don't see any particular problem, especially if it does exactly the same thing under Ubuntu :'(
I confess I don't understand why only this file is a problem...

Can you help me with 3 other questions please:
  1. How big is the `boot.py` you are trying to send?
  2. If you rename your local `boot.py` to another name (`toto.py` for example) and send it, is there still an error?
  3. If you send another test file named `boot.py` instead, does it work?

Not easy to diagnose, I'm curious to understand!

jczic, 2023-04-11:
Did you find anything?

nio101, 2023-04-12:
Yes, it seems to be size-related.
Original file was 3.2kB.
I managed to upload a tiny boot.py file, and I had the same error trying to send another big file (not using the boot.py filename).

jczic, 2023-04-12:
@nio101, you may not have enough space in the flash memory to write a large file.
Have you had time to see the progress bar?
No one seems to have this problem and multi-meg files can get through (on SD card as well) depending on the space available.