jczic/ESP32-MPY-Jama #38: System information not reporting SPI ram info ESP32-S3

ThinkTransit, 2023-04-12:
Hi,

Great software!

I noticed when using an ESP32-S3 32MB flash and 8MB RAM, the system information says SPIRAM available no, should that be yes?

[image]

jczic, 2023-04-12:
Hi @ThinkTransit,
Ok I see, the term of spiram does not appear in the implementation.
I checked for your GENERIC_S3 type card and indeed, the configuration file does not directly include the use of SPIRAM so that must be why. I'm only doing this check for now...
( https://github.com/micropython/micropython/blob/master/ports/esp32/boards/GENERIC_S3/sdkconfig.board )

ThinkTransit, 2023-04-12:
Ok that makes sense. Maybe it might be useful to add total memory to the system information, gc.mem_alloc() + gc.mem_free().

Just a suggestion.

jczic, 2023-04-12:
Exact! However there is at the bottom left the gauge with these 2 informations but all the fonts seem to be unstaggered on your system :(

anglerfish27, 2023-09-16:
Agree I have the same issue. On an Unexpected Maker ProS3 I confirmed it has 8MB of RAM, the GUI says there is no SPI RAM (maybe because its used?) and other msgs are wrong too. But I understand its a new board.

On an official devkit s3-c1 (N8R8) it says "flash memory size 8MB" and SPI RAM available Yes. However when running a program I have about 62K of RAM only.  I am running the correct and latest firmware for octal SPI.