jczic/ESP32-MPY-Jama #39: repl issues on linux (ubuntu)

ThinkTransit, 2023-04-12:
Hi,

The repl display does not appear to show underscores _ maybe this is a problem on my machine.

Also the repl does not accept input when there is no >>> prompt. For example if you try import webrepl_setup it is impossible to complete the setup, repl is not recognising input.

[image]

jczic, 2023-04-12:
Hello @ThinkTransit and thank you for using 🙂

I didn't understand the problem with the underscores, they work like any other characters though.

Concerning the "Input" mode of the REPL, in fact, the terminal does not support it because it is not directly linked to the same REPL as the default one of the serial UART but to the raw mode...

ThinkTransit, 2023-04-12:
Hi @jczic,

Sorry I should have been clearer. What I meant is that the underscores aren't visible in the repl, until after you press enter.

It's a very strange issue, maybe it is a problem with my computer. See here.
[image]

jczic, 2023-04-12:
Oh yes, it's just the fonts that are out of sync on your system apparently :(
Are you on Linux?

jczic, 2023-04-12:
Let's say no, there is only one font that is specifically charged from this folder:
https://github.com/jczic/ESP32-MPY-Jama/tree/master/src/content/ressources/font