jczic/ESP32-MPY-Jama #43: Jama Func microSD test function

ghvau, 2023-05-05:
[jama-func-sd.zip](https://github.com/jczic/ESP32-MPY-Jama/files/11408823/jama-func-sd.zip)

I have problems the find the right connection data of the "SD Card" function in the main menu.
Therefore I build a "quick and dirty" test for the Jama Funcs with some good segments from github.
Maybe you are interested.

jczic, 2023-05-05:
Hey, this is really cool! :)

You didn't have the **SDCard** library in your MicroPython port right ?

So here you can actually choose the SPI GPIOs for testing the particular ESP32 boards.
But indeed, the SD Card menu in the software doesn't work for you I see, what exact message do you get when you try to activate it from this page ?

Also, in the information of the "Jama Func", you can put your mail (see the template) and your site or github in www :)
I could add it to the repository without problem then with an info in the main readme!

ghvau, 2023-05-06:
Hi,

when I try to start the SD Card funktion I get the error "No SD Card 
present".

The SDCard library is available, but (till now) I can't run it for my 
boards.

I try to find the position in your prog. where you set or I can change 
some parameter.

But I see also that there are preffered slots (card wiring) and none of 
them hit my wiring.

If you like the JamFunc feel free to use/modify it,

I am be silent in the background (no active repository).

jczic, 2023-05-06:
Hi @ghvau, in fact, the SD Card part of the program simply uses the default SD Card slot of the ESP32 which works on most boards.
You have indeed particular boards that require different GPIOs.

However, if you get this error message, it means that the library is indeed supported on your MicroPython and in that case, it should be usable with your boards.
Look at all the options available here :
https://docs.micropython.org/en/latest/library/machine.SDCard.html

You could use it by proposing to configure these options instead of integrating another more restricted library in the Jama Func?

At what level did you manage to use this library? (there is the SLOT option which can be important too).

It would be a pleasure to add your code :)

ghvau, 2023-05-07:
[SDCard_var-native.zip](https://github.com/jczic/ESP32-MPY-Jama/files/11415705/SDCard_var-native.zip)

Hi @jczic,

following your mind, I try the "native" SDCard Class, w/o any additional libs.
The result you find in the attachment, as minimalist implementation.

I thing the native solution is not usable for each board config (exotic card wiring).
Sometimes after repeated call of "sd = SDCard(slot=x ... "  the function hang and 
my integrated reset() get a trigger. But principle it works.

jczic, 2023-05-07:
Hello, yes I think it is more logical to use the built-in library and why not test if it's supported during the `import` instruction.

In fact, if sometimes the configuration crashes the board, it's simply because the selected GPIOSs are already used for something else and that causes a hardware error on the board.
Do you think it is really necessary to reboot in case of a problem?

Also, slots 0 and 1 are, for the ESP32, the default slots without indication of GPIOs. Only slots 2 and 3 can be set via an SPI interface.
Also, you may have a problem with the bus size in bits (argument `width`).

Your SD Card test is an excellent idea and actually, does it work well on your exotic cards?
(eventually, you should delete the file created at the end of the test?)