jczic/ESP32-MPY-Jama #45: Unreliable connection to Lolin S3 Pro

ejeklint, 2023-05-24:
I have big problems getting MPY-Jama to work properly with a Lolin S3 Pro card. When uploading a file I usually, but not all the times, get a "Timeout...", and after that I get "An error has occurred", or "A process is in execution", or "A process is already in execution".

Disconnect/reconnect + restarting MPY-Jama can sometimes get the setup running again, but only after a few attempts and after one or two edits of main.py (running a simple blinky application) the problems are back. Also: ctrl-commands in the repl terminal does nothing. The card is preloaded with micropython and seem to work as it should. I've tested on two machines (macOS 13.4) with the same result. I haven't got a clue on how move forward with this... 😕 

[image]

jczic, 2023-05-24:
Hi @ejeklint, do you use the binary application for the software or do you run it from the repository sources?
In which case, could you try the same operation by changing the timeout at line **616** of the `src/esp32Controller.py` file by putting for example *timeoutSec=10*?
3 seconds is supposed to be enough to write the file buffers but maybe there is sometimes a latency on your card (which I don't have to test...).

ejeklint, 2023-05-24:
I'm running the downloaded binary application, on an M2 Macbook pro. I have very little experience with Python but will give it a try, building from source. *gulp*

ejeklint, 2023-05-24:
Nope, I'm not python-savvy enough to build a new binary. Giving up for the moment.