jczic/ESP32-MPY-Jama #47: Are the ESP32-C3 boards supported

deepcoder, 2023-06-05:
I am not able to connect a ESP32-C3 board. When trying to connect one, I get the following message:

```
The device on the "/dev/cu.wchusbserial54F40144441" port is not compatible.
```

Is there a list of the supported board types in the documentation somewhere?

Thx!

jmbruges, 2023-07-22:
It works for me with a C3 board. 

Are you using version 1.2.0?

anglerfish27, 2023-09-16:
I'm having issues, it partially detects my Unexpected Maker ProS3 board. I am using the latest stable MicroPython firmware. Only some of the info the GUI is right. Of course no Pinout for it either. It's a really popular board recommend testing and adding to the GUI. I really hope your program continues to be updated. Its AWESOME so long as it its a chip it knows. Please continue to work on it!!! I love it!

Also I have an EspressIF (from the company) dev board ESP-32-DevKit-C1 (N8R8) which seems to display mostly ok. However I see it is only showing a dismal 62Kb of RAM available. How to "activate" the Octal SPI as RAM? On my other Unexpected Maker ProS3 it was already setup and I have a whopping 8MB of RAM. I want to do the same with this official dev kit. Can that be done via your GUI? or another way? I have the latest stable MP loaded on it and yes its the correct one for octal SPI. Feels like the chip is sitting there not doing anything and I want to use it for RAM! Websockets and large images will not suffice on a meager 62kb RAM. Thoughts?

Respectfully,
Anglerfish27

anglerfish27, 2023-09-16:
I wanted update my post so that the right information is shared. Instead of using the latest firmware I erased and installed the stable build on the MP site. That worked. The SPI RAM is now detected as in use as 8MB RAM/Heap! The erase portion of your GUI worked and it appeared to flash the .bin correctly, however when I tried to load thonny it was faling with FFFFFF over and over being spewed. So I ended up using the instructions on the MP site to erase and flash the stable build using esptool.py after that everything works. So I retract that part of my post above. Apologies. I love that you can save configs such as the Wifi + Wifi AP. Awesome stuff. Keep it coming!!

Cheers
Anglerfish27