jczic/ESP32-MPY-Jama #52: Unable to connect to device on Macbook M1 Pro: Port is not compatible

shahrukh-alizai, 2023-08-15:
**Screenshot:**

I'm getting this error on Macbook M1 Pro. 

[image]

However, it works fine using **esptool** on terminal.

[image]

ghvau, 2023-08-16:
In your installation the error is correct -> because esp32 (-> tool name) is unequal esp8266 !

https://github.com/micropython/micropython/issues/2709
"... in Built-in Functions there is a **'compile()'** function that **hasn't been implemented on the esp8266**..."

Einariux, 2023-09-04:
Same problem with ESP32 Nodemcu,but firmware flash and erase.MacbookAir M1

FabianClemenz, 2024-01-12:
I have the same Problem with my M2 Pro Macbook.

I was able to flash the firmware with MPY-Jama and with esptool directly. But it can't connect to the board.

[image]
[image]

I'm using the WT32-ETH0 Board over a USB-to-TTL Connector

marksull, 2024-02-07:
Maybe this will help others if you see the same issue and might help lead to a root cause. On the same M2 with the same ESP32-S3:
1) If I clone this repo and start the app from the repo, I get the "port is not compatible" error when trying to connect to the ESP32-S3
2) If I download the pre-built MacOS app and run it, I can communicate with the ESP32-S3 fine

I can switch between the two installation styles, and the issue is consistent. I suspect a MacOS security issue is causing the problem, but looking through the console logs I couldn't identify a reason.

mwalimu, 2024-10-07:
I found that in order to get the board to connect it had to have micropython already installed on the device.