jczic/ESP32-MPY-Jama #59: Colour theme

GerardHarkemaAvans, 2024-05-22:
Dear developer, I think this is one of the better IDEs for developing in MicroPyhon. Unfortunately, I have poor eyesight and cannot read texts with limited contrast, for example red text on a black background. In the future can you add the options for choosing a color theme? Thank you in advance!

jczic, 2024-08-16:
Hi and thank you @GerardHarkemaAvans  ! ♥

I'm sorry, but unfortunately I'm not working on it at the moment :(
I'm co-founding an AI start-up dedicated to e-commerce customer service... ([www.achille.ai](http://www.achille.ai/)).

However, you can always open the package/zip and go to the CSS file to try and change the fonts and sizes.
It's Web format behind the UX ;)

Kongduino, 2024-09-06:
So, I suppose no light mode either?